eirikpre/VSCode-SystemVerilog #8: Add syntax highlighting for compiler directives

mrvkino, 2018-02-07:
Could it be possible to add syntax highlighting rules for the compiler directives and Macro:
```
`__FILE__
`__LINE__
`begin_keywords
`celldefine
`default_nettype
`define
`else
`elsif
`end_keywords
`endcelldefine
`endif
`ifdef
`ifndef
`include
`line
`nounconnected_drive
`pragma
`resetall
`timescale
`unconnected_drive
`undef
`undefineall
```
See section 22 of IEEE 1800-2012 SystemVerilog.

eirikpre, 2018-02-07:
Good idea, will add!
`define | ifdef | ifndef | else | endif` is already in.
They are purple ish in the Dark+ theme equal to for|while|else|if etc( keyword.control ).
Do you have a comment on the color (scope), would normal blue maybe be better? 
The other `'tick'xxx` are matched as constants (light green in dark+). I'm open to suggestions on the as well, but I deemed it most natural since they're often used as constants/paths.

closed 2018-02-08 by eirikpre

mrvkino, 2018-02-08:
Ok, it's just that the colouring now is confusing/not distinctive enough in my opinion.

In the example bellow, I would of expected the localparam PARAM_VALUE and the preprocessor TUTU colour to be different (since they are different type of element). Also, the TUTU and TOTO pre-processor would of been the same colour. I understand the logic to treat pre-processor as constant, but in my mind they are something different from regular systemverilog.

Also, the `ifdef being the same colour as the 'if' bring difficulty to distinguish between systemverilog code and pre-processor function.

[image]

I view the pre-processor as their own little separate language, since they have their own interpretor that run separately from the systemverilog (aka replacing manual copy/paste no feasible with generate)

I use the built-in solarized dark theme, but that doesn't matter, the syntax rule seems consistent between the themes, except for the begin..end that have the same colour as the builtin.

eirikpre, 2018-02-08:
Yeah, I agree with you. 
Problem being that some of the themes do not cover/distinguish the different scopes.
I suppose I can try with the the function color ( scope=support.function... ). You would most likely use comp-directives mostly in synthesizeable RTL, so there wouldn't be much other functions taking up attention.
Patch with the change incoming shortly.

mrvkino, 2018-02-08:
I looked how VSCode managed the preprocessor directive in C. Do you think aligning with their choices could help?

They seem to use the meta scope for the macro and inside they go as usual with the variable, punctuation but with a sub-scope of preprocessor. The following example is from [here](https://github.com/Microsoft/vscode/blob/master/extensions/cpp/syntaxes/c.json)

```
{
			"begin": "(?x)\n^\\s* ((\\#)\\s*define) \\s+    # define\n((?<id>[a-zA-Z_$][\\w$]*))      # macro name\n(?:\n  (\\()\n    (\n      \\s* \\g<id> \\s*         # first argument\n      ((,) \\s* \\g<id> \\s*)*  # additional arguments\n      (?:\\.\\.\\.)?            # varargs ellipsis?\n    )\n  (\\))\n)?",
			"beginCaptures": {
				"1": {
					"name": "keyword.control.directive.define.c"
				},
				"2": {
					"name": "punctuation.definition.directive.c"
				},
				"3": {
					"name": "entity.name.function.preprocessor.c"
				},
				"5": {
					"name": "punctuation.definition.parameters.begin.c"
				},
				"6": {
					"name": "variable.parameter.preprocessor.c"
				},
				"8": {
					"name": "punctuation.separator.parameters.c"
				},
				"9": {
					"name": "punctuation.definition.parameters.end.c"
				}
			},
			"end": "(?=(?://|/\\*))|(?<!\\\\)(?=\\n)",
			"name": "meta.preprocessor.macro.c",
			"patterns": [
				{
					"include": "#preprocessor-rule-define-line-contents"
				}
			]
		},
```

eirikpre, 2018-02-09:
Mhm, kinda hard to say, without creating my own theme I actually have no control over how the actual textmate scopes are used, I relate to the TM spec [here ](https://manual.macromates.com/en/language_grammars) for the most part.
But! I looked into the source of Dark+ theme, and I see they give a red color to punctuation.definition.group.regexp, might be a hacky solution to use that. However, I have no idea of what other themes do.
I'll not spend too much more time on this, but I'll try the regex color in the next release.
Currently working on outline for current document & outline for the module hierarchy (but that might take a while longer)

commit 525f166f (2018-02-08): Added all compiler directives, fixes #8, published patch 0.2.5