eirikpre/VSCode-SystemVerilog #13: Large indexing files freeze vscode

alea75, 2018-04-23:
When I add a new folder to the workspace and if this folder contains large .sv/.v files (or a big amount of files) then vscode is freezed until a pop-up appears saying "The window is no longer responding". Then I have choice between "Reopen", "Keep Waiting" or "Close" vscode.

I have tried "Keep Waiting" but I never had vscode that works again (new pop-up windows at each time).

eirikpre, 2018-04-23:
Darn, I was afraid that might happen. I'm not throttling the cpu/ram usage, just counting on vscode / os to keep it stable. 
It has worked okay in my workspace with 15k modules, but the current solution is not really scaleable.

I could perhaps limit it to opening 100 files at once, or something like that.

eirikpre, 2018-05-23:
It is throttled now with 100 files at time. It's relatively static to computer specs, so slower computers with less ram might have a problem. I'm unsure how to throttle it based on specs / speed.
100 files at a time uses 2G RAM during indexing for my workspace of around 10k .v/.sv files

closed 2018-05-23 by eirikpre