eirikpre/VSCode-SystemVerilog #14: Disable indexing

aviadb, 2018-05-23:
I am working on a very large remote repository
with more than 700k .v,.sv files.

Indexing starts automatically and hogs the cpu, memory and network.
An option to disable indexing or index only certain folders under the workspace would be great.

In addition,
Since some files may appear under different locations,
"Go to definition" opens the first occurrence, not showing all possible definitions.

eirikpre, 2018-05-23:
That, indeed, is a huge workspace!

I have absolutely been thinking of a way to disable the indexing, I just haven't got around to implement it yet. I understand that it's very much a mandatory feature, and it'll be the feature for the next work session for sure :)

Yeah, the Go to definition was quickly assembled, and I didn't see a built in/API way of adding multiple results, but I assume it is possible.

aviadb, 2018-05-23:
The c/c++ extension provided by vscode developers offers
got to definition with multiple occurences
[](https://github.com/Microsoft/vscode-cpptools)