eirikpre/VSCode-SystemVerilog #18: support for verilog file

wl6g11, 2018-07-12:
The feature of **Go to definition** is not working for the verilog file (*.v).
Is anyone can fix it?

eirikpre, 2018-07-12:
Hm, haven't noticed that before. Go to definition uses the indexed modules, which should consist of modules that reside in both *.v and *.sv files (within your workspace).
You can check your indexed modules with ctrl+shift+p -> go to symbol in workspace.

It might be that another extension takes precedence over mine for verilog files, and not SystemVerilog files?

wl6g11, 2018-07-13:
When a *.sv file is opened, I can go to the definition of a instance module in this *.sv file.
If a *.v file is opened, I can not go to the definition of a instance module in this *.v file.
The indexed modules work fine for both *.v and *.sv files since the go to symbol feature works fine with both *.sv and *.v file.
It seems that the go to definition feature is not enable for the *.v file.
I found that the code `selector:vscode.DocumentSelector = { scheme: 'file', language: 'systemverilog' }` in extension.ts.
Does this code means that the feature only available for systemerilog?

eirikpre, 2018-07-13:
You are correct, that one should probably contain verilog! 
But.. Unfortunately, I haven't had time to implement go to definition for symbols in the document (yet). And I won't be able to in at least the coming week since I'm on holiday. 
In other words, go to definition and go to symbols in document are not connected for *.v nor *.sv file(yet) :(

wl6g11, 2018-07-16:
I created a new pull request for this problem.
This is a very nice extension, thanks for you job.
Hope you can release a new version, then I do not need to rename all my *.v files to *.sv in order to make the go to definition feature works correctly.

eirikpre, 2018-07-16:
Awesome! I'll review it when I'm not limited to mobile. In the meantime, I think you can register *.v files as systemverilog in user-settings -> files.associations

eirikpre, 2018-07-31:
Published a patch with the latest PR included

closed 2018-07-31 by eirikpre