eirikpre/VSCode-SystemVerilog #21: Go to definition

CatWithoutBoots, 2018-07-31:
Can you add a capatibility for work with "Go to" for tasks and functions?

eirikpre, 2018-08-03:
This is something I have wanted to add, but there hasn't really been time.

At the moment I only see it being able to Go to tasks/functions within the current file.

eirikpre, 2019-01-15:
*Update
> At the moment I only see it being able to Go to tasks/functions within the current file.
has been working since v0.5, at the very least.
Global go to definition, not yet

eirikpre, 2019-08-23:
Heya, I would previously not add this because indexation in big workspaces would take forever.

But it is now aware of workspace size, so if you have:
less than `settings.parallelProcessing*1000` it will index all global tasks/events/typedef++.
less than `settings.parallelProcessing*100` it will index all global tasks/events/typedef++ including one hierarchy inside found global signals(eg file:class->function, instead of just file:function)

eirikpre, 2019-08-23:
edit: less than `settings.parallelProcessing*XXXX` as in number of `**/*.{v,sv} files in workspace

closed 2019-08-23 by eirikpre