eirikpre/VSCode-SystemVerilog #23: Parameterized declarations mess up syntax highlighting

djoffe, 2018-08-13:
In the following code example, lines 2 and 3 (the uvm_analysis_export declarations) mess up the lines following them. Comment those two lines and the rest of the code looks fine.
- Uncomment one line and the syntax highlighting is messed up until the first `new` keyword following it (`function new`).
- Uncomment the second line and the syntax highlighting is messed up until the second `new` keyword following it (`super.new...`).
- And so on for each parameterized declaration (The `#` seems to be causing the issue here)

```sv
class env extends uvm_env;
    uvm_analysis_export#(some_pkg::item)    some_export;
    uvm_analysis_export#(some_pkg::item)    some_other_export;

    // References
    reg_bus_uvc_pkg::env   reg_bus_env;
    reg_bus_uvc_pkg::agent reg_bus_agt;

    function new(string name, uvm_component parent);
        super.new(name, parent);

        some_export         = new("some_export", this);
        some_other_export   = new("some_other_export", this);
    endfunction
endclass
```

[image]
[image]
[image]

djoffe, 2018-08-15:
Correction, it is not the `new` keyword that closes the bad highlighting area, but any opening paren `(`.

eirikpre, 2018-08-16:
Yes, I know this is an issue. Also when you have extra paranthesis in a portlist it doesn't match the number of parantheses.
It has been a while since I last tinkered with the textmate setup. I was kinda hoping we could have only one source for the syntax highlighting.
I'll fix this the next time I'm working with the extension :)

eirikpre, 2018-09-18:
Fixed with 0.4.1 (released now, thanks for reporting)

commit 19393826 (2018-09-18): 0.4.1 preparations, fixes #23
closed 2018-09-18 by eirikpre