eirikpre/VSCode-SystemVerilog #29: SV types not indexed

rahulthom, 2018-09-13:
First let me say, thank you creating this SV language support for VSCode - extremely helpful.

I'm having an issue with indexing, specifically for types. Indexing of classes seem to be OK - i.e. if I click on an instance of a class, I can go to the definition of the class.

For types however, when I right click on a type and hit go to definition, it doesn't appear to work - I get 'No definition found'. A simple example is this:

example_types.sv
- contains type example_t

test.sv:
- contains:  example_t example;
  -> if I try to find the definition of example_t it can't find it

Would it be helpful to also include types in the indexing.

eirikpre, 2018-09-14:
Hi thanks!
Yeah, that depends on what kind of type you are looking for. It is only extracting the following elements from files in the workspace: module/class/interface/package/program. I do not want to look for too much, as that'll make the indexing slower. What type did you have in mind or is it typedef?
Here's the regex used on each .sv/.v/.svh/.vh: 
_/^\s*(?:virtual\s+(?=class))?(module|class|interface|package|program(?:\s+automatic)?)\s+(\w+)/_

I've picked out the words above because those are most likely to be used 'globally'. :)

rahulthom, 2018-09-14:
Yes, typedef is what I'm interested in with indexing. 
Most of the time it's a 'typedef enum ...', but my projects also use simpler typedefs like typedef int, etc.

Would typedef support for indexing be something you would consider?

Thanks,
Rahul

jiaxiang-liu, 2019-06-07:
Agree with @rahulthom , if it supports typedef will be excellent.

eirikpre, 2019-08-09:
Heya, long time no see!

I added typedef indexing for workspaces that are not too large.

They are indexed as long as `settings/systemverilog.parallelProcessing`*1000 > number of .v/.sv files in the workspace

Enjoy :)

closed 2019-08-09 by eirikpre