eirikpre/VSCode-SystemVerilog #31: linter capability

yjohner, 2018-10-11:
First thanks for this extension I like it.
It could be interesting to add linter capability like in "Verilog HDL support for VSCode"
As I'm Mentor tool based my preference would go there first but..

eirikpre, 2018-11-14:
Hm this is a hard one.
It's absolutely possible to create some regexes resembling some linter rules, but I don't think I have the time to create something that'll cover even 50% of all the strange functionality in SystemVerilog.
The other approach is to interface with some open-source/commercial linter, then I need to learn an API and make an adapter in the extension to the API, which is likely a lot of work too. (Also require users to install external tool not part of npm)

I'll check out what they do in the extension mentioned.

markusdd, 2021-02-14:
Frankly, with verilator there is no need to implement something at all.

It's lint-only function is actually pretty comprehensive and it can flag all kinds of things already.

joecrop, 2021-08-19:
> Frankly, with verilator there is no need to implement something at all.

IMHO, Verilator works OK for Verilog and most SystemVerilog RTL, but it totally falls apart when you have verification code like classes, UVM, external references, and most importantly large projects with long compile times.

I have been experimenting with `verible-verilog-lint`. It can run on a single file without the need to compile any other files, and does a half-decent job finding issues. It also supports all SystemVerlog syntax.

I would like to contribute this feature. @eirikpre would you be open to this? Would you suggest making a category for lint, separate from compile, or just adding it as another compile option?

eirikpre, 2021-08-23:
I'm very open to all kinds of feedback, but I haven't been able to spend any time on the extension the last year (or almost two). 
I do think that the extension require a proper cleanup to support future development, to make it more modular, and from there be able to make each supported feature more stable and consistent (in addition to new big features, such as completionProvider). 

I too have looked into `verible` for linting (and more), but at the time I investigated macros where not supported, which I think are popular among many.

joecrop, 2021-08-24:
Yes, that's true. When I write test cases like this, `verible` totally bombs with a boat load of errors.

```
`TEST_BEGIN

int x = 4;
drive_dut(x);

`TEST_END
```

It seems to me that the best option would be to allow the user to exclude certain files from linting.

joecrop, 2021-10-13:
@eirikpre are you able to provide feedback on my MR?