eirikpre/VSCode-SystemVerilog #34: High CPU usage

itaygr, 2018-12-12:
I'm experiencing high CPU usage which makes vscode very unresponsive.

In my case, there were about 1000 files indexed, but the sluggishness is seen *after* the indexing has finished.

It's slow even when just using the mouse to select text. (not just when doing symbol lookup or auto-complete etc).

Is there a way to find out what the extension is doing while it's running slowly?

(Disabling the extension makes vscode responsive again)

eirikpre, 2018-12-13:
Hi!

Yes, that might be because the indexation is trying process files in parallel to save time.  Currently it's limited to only open 100 files at once, but for machines with low RAM it seems to struggle some. I'm in a the process of exporting that # number of files in parallel as an user setting, so it can be lowered (or increased).
I have no idea how to scale it according to the machine running it.

However, the unresponsiveness should stop after it's done indexing. ( Uses a few minutes to index around 10k files, and only a few seconds on ~1k files on my work computer )

itaygr, 2018-12-13:
Thanks for your reply.

Yeah... I'm pretty sure indexing finished when the slowness is happening.

Anything i can look at?

eirikpre, 2018-12-13:
Hm, hard to say, not so sure how to attack this :(
You could try to look in 'Developer: Open Process Explorer' to see what is going on.
My extensionHost is currently using 0% CPU and roughly 550mb RAM after indexing 12k objects.
Also try without any other extensions, there might be a conflict (I'm seeing the python ext is doing something weird sometimes)

itaygr, 2018-12-13:
good suggestions.

I've disabled Python in the workspace, but that didn't help.
What process explorer shows is:
0% CPU and 577MB on the extensionHost
0% CPU on Window

I then select some text using the mouse, and i can see the "window" process going up to 14% or so (8 core system with 32GB ram). The text selection isn't smooth, it's stuttering. 

i'll keep playing around with this and let you know if i find something...

itaygr, 2018-12-14:
I dug in a bit further, managed to see this:

```
SyntaxError: Invalid regular expression:
**<My Entire Source code file was here!>**
/: Range out of order in character class
	at new RegExp (<anonymous>)
	at SystemVerilogWorkspaceSymbolProvider.provideWorkspaceSymbols (.../.vscode/extensions/eirikpre.systemverilog-0.5.2/out/providers/WorkspaceSymbolProvider.js:30:27)
	at docSymProvider.provideDocumentSymbols.then.docSyms (.../.vscode/extensions/eirikpre.systemverilog-0.5.2/out/providers/HoverProvider.js:26:50)
	at <anonymous>
e.$onExtensionRuntimeError @ mainThreadExtensionService.ts:43
```

So for some reason the extension is using my entire source file as a 'regex'. That could explain the slowness... especially as it's throwing an exception etc.

What do you think?

itaygr, 2018-12-14:
Another thing i noticed was:
```mouseHandler.ts:89 [Violation] Handling of 'mousewheel' input event was delayed for 453 ms due to main thread being busy. Consider marking event handler as 'passive' to make the page more responsive.
```

and:
```dom.ts:206 [Violation] Added non-passive event listener to a scroll-blocking 'mousewheel' event. Consider marking event handler as 'passive' to make the page more responsive. See https://www.chromestatus.com/feature/5745543795965952
e @ dom.ts:206
h @ dom.ts:224
t._setListeningToMouseWheel @ scrollableElement.ts:315
t @ scrollableElement.ts:214
t @ scrollableElement.ts:497
t @ editorScrollbar.ts:53
t.createViewParts @ viewImpl.ts:159
t @ viewImpl.ts:131
t._createView @ codeEditorWidget.ts:1461
t._attachModel @ codeEditorWidget.ts:1359
t.setModel @ codeEditorWidget.ts:413
(anonymous) @ textFileEditor.ts:137
then @ winjs.base.js:1587
(anonymous) @ textFileEditor.ts:119
then @ winjs.base.js:1587
t.setInput @ textFileEditor.ts:118
t.doSetInput @ editorControl.ts:175
t.openEditor @ editorControl.ts:67
t.doShowEditor @ editorGroupView.ts:780
t.doOpenEditor @ editorGroupView.ts:770
t.openEditor @ editorGroupView.ts:739
r @ tabsTitleControl.ts:465
(anonymous) @ tabsTitleControl.ts:477
```

Not sure if your extension adds this non-passive listener...

patstew, 2019-01-08:
I'm also getting random slowdowns which I think is down to this extension, even after indexing. I've fixed the regex issue above in #35, not sure if that's solved the problem yet.

itaygr, 2019-01-10:
Thanks @patstew , i've pulled the latest code and running the extension with your change now. Will update if it helps/doesn't.

eirikpre, 2019-01-14:
Thanks for help, these things are hard to debug. @patstew 's PR is now published, together with a setting named parallelProcessing, where you can tune how aggressive the indexation will be done. Lower number will be safer but slower. 
I don't think the extension adds non-passive listeners, but it might be that the new hover is converted to a non-passive listener, perhaps there should be a setting to disable it.

itaygr, 2019-01-21:
Just wanted to follow up, unfortunately the slow downs are still unresolved..

I'm not really sure how to debug this. My best case it's to do with 'Style Recalculation', based on some performance recordings i took while i was stuttering.

eirikpre, 2019-03-13:
Hey!
Sorry for not following up lately. 

I just made some modifications that I hope that will make it more stable, at the very least it will be a lot faster now. In my workspace it went from about 2 mins to 1 min.

I think that if I can replace this **[for-loop](https://github.com/eirikpre/VSCode-SystemVerilog/blob/master/src/providers/WorkspaceSymbolProvider.ts#L82)**
with a map of promises, that will make it responsive while indexing. (The for loop stalls the GUI i think)

itaygr, 2019-03-13:
@eirikpre That's great - i'll give it a try!

Most of my stalls are after indexing. Is this loop only needed while indexing? Maybe it can be bypassed after that is finished.

eirikpre, 2019-03-13:
Yes, it's only used while indexing. 
I do want to experiment with writing the indexed modules to a file, instead of keeping them in memory.
That may solve stalls post indexing.

itaygr, 2019-03-14:
Yeah, looks like the slow-downs are still here.

If you want help experimenting, I can try out versions from a branch you make

eirikpre, 2019-03-20:
A damn, that's sorry to hear.
You could try to clone the repository and run while debugging.

Install node
Open the clone in vscode(open folder/workspace)
npm install in the terminal
press f5 (start debug)
try add some breakpoints in my code (ideally it should not be running any of the extension's code automatically after indexing, unless asked for it)

eirikpre, 2019-03-20:
You could also fork if you want to make some edits and PR, up to you :)

eirikpre, 2020-05-06:
I consider this as good as it can be at this point. The greedy parallelism can be tweaked, and the extension now feature to save the workspace state(so it wont index already indexed workspaces). Indexing can also be disabled on open, and then triggered manually by command or statusbar.

For proper background job, future LSP have to handle indexation gracefully.

closed 2020-05-06 by eirikpre

itaygr, 2020-05-06:
Thank you for you hard work, I use this extension daily for my work and it's great!