eirikpre/VSCode-SystemVerilog #37: Version 0.5.3 cause settings file to not open

mrvkino, 2019-01-14:
Since version 0.5.3 the settings file is not able to open with the following message:
```
Unable to open 'Settings': ((i.description || i.markdownDescription) || "").split is not a function.
```
This behaviour disappear if I install version 0.5.2 or uninstall the extension. I found it by disabling all extension and it seems it is the only one causing this behaviour.

I am on:
VSCode: 1.30.2
OS: MacOS 10.14.2
Version of the extension: 0.5.3

eirikpre, 2019-01-14:
Oh noes, it worked locally, but I didn't have time to test separately. Will
fix tomorrow!

eirikpre, 2019-01-15:
Fixed with version 0.5.5
I used an array for the description of the new setting instead of pure string. Now it's a long single string.

closed 2019-01-15 by eirikpre