eirikpre/VSCode-SystemVerilog #39: Extension host terminated unexpectedly, potential file listener/memory leak?

tovine, 2019-03-11:
I think this problem may be caused by this extension, see debugger log below:

```
<--- Last few GCs --->

[3408:000001C003C71500]   610787 ms: Mark-sweep 3519.9 (3668.2) -> 3519.9 (3668.2) MB, 2946.7 / 0.0 ms  allocation failure GC in old space requested
[3408:000001C003C71500]   613882 ms: Mark-sweep 3519.9 (3668.2) -> 3519.9 (3668.2) MB, 2990.2 / 0.0 ms  last resort GC in old space requested
[3408:000001C003C71500]   616978 ms: Mark-sweep 3519.9 (3668.2) -> 3519.9 (3668.2) MB, 3095.6 / 0.0 ms  last resort GC in old space requested

<--- JS stacktrace --->

==== JS stack trace =========================================

    0: ExitFrame [pc: 0000016F59184481]
Security context: 000001A781625709 <JSObject>
    1: lineAt [c:\Program Files\Microsoft VS Code\resources\app\out\vs\workbench\services\extensions\node\extensionHostProcess.js:~341] [pc=0000016F5942922D](this=0000038CAF4022E1 <Object map = 0000036C29AD8061>,t=119318)
    2: /* anonymous */(aka /* anonymous */) [C:\Users\tovine\.vscode\extensions\eirikpre.systemverilog-0.5.6\out\providers\WorkspaceSymb...
```
This was also in the log prior to the above stack trace:
```
[dd5] potential listener LEAK detected, having 2849 listeners already. MOST frequent listener (2846):
check @ event.ts:451
event.ts:452     at e.registerListeners (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4160:144)
    at new e (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4159:623)
    at t.create (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:217:102)
    at s._createInstance (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:1381:680)
    at s.createInstance (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:1381:92)
    at h.loadOrCreate (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4184:222)
    at m.createReferencedObject (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4527:239)
    at m.acquire (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:67:172)
    at Proxy._createModelReference (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4528:955)
    at Proxy.createModelReference (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4528:499)
    at f._handleAsResourceInput (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4268:557)
    at f.$tryOpenDocument (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4268:37)
    at h._doInvokeHandler (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:3236:459)
    at h._invokeHandler (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:3236:151)
    at h._receiveRequest (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:3234:825)
    at h._receiveOneMessage (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:3233:707)
    at _protocol.onMessage.e (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:3231:990)
    at d.fire (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:73:416)
    at r (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:314:233)
    at e (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:314:279)
    at d.fire (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:73:416)
    at o (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:315:556)
    at Socket._socketDataListener.e (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:315:762)
    at Socket.emit (events.js:182:13)
    at addChunk (_stream_readable.js:279:12)
    at readableAddChunk (_stream_readable.js:264:11)
    at Socket.Readable.push (_stream_readable.js:219:10)
    at Pipe.onread (net.js:636:20)
check @ event.ts:452
event.ts:451 [2h7] potential listener LEAK detected, having 2794 listeners already. MOST frequent listener (2791):
check @ event.ts:451
event.ts:452     at new a (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:1477:992)
    at m.createByFilepathOrFirstLine (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:1479:753)
    at e.getOrCreateMode (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4165:10)
    at e.createTextEditorModel (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:2377:828)
    at createTextEditorModelPromise.doLoadBackup.then.i (file:///C:/Program Files/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4164:460)
    at process._tickCallback (internal/process/next_tick.js:68:7)
check @ event.ts:452
extensionHost.ts:234 Extension Host
extensionHost.ts:235 Debugger attached.

extensionHost.ts:234 Extension Host
extensionHost.ts:235 FATAL ERROR: CALL_AND_RETRY_LAST Allocation failed - JavaScript heap out of memory
```

eirikpre, 2019-03-11:
Yep, trying to fix it.
Came with the newest update of Code.
If it would be possible to give the process more Ram, that might help

eirikpre, 2019-03-11:
For other people with the problem:
Subscribe to this thread, and disable extension/downgrade until closed

eirikpre, 2019-03-13:
I believe this is fixed with v0.6 ( now released )
It is now indexing a lot faster as well + fuzzy matching the results.

closed 2019-03-13 by eirikpre