eirikpre/VSCode-SystemVerilog #40: LSP Server

DeeeeLAN, 2019-03-18:
I just want to make sure I am reading correctly. Currently, this extension only works in VSCode, not as a full LSP server, usable with any client, correct? That is what the Futuer to do means by "Back-end Language server for Systemverilog"? Has work already started on this feature, or is it still untouched?

eirikpre, 2019-03-20:
Still untouched.

I think that a proper LSP server is a bit too much work to implement as a hobby. If it would be possible to wrap around an open-source compilator, that would be less daunting.
I don't think it would be a good idea to attempt to create another compilator :) 

The current matching and LSP reside as part of the extension host, which I think works pretty good, but one has to use Typescript to develop.

mithro, 2020-01-28:
FYI @dalance has a System Verilog Language server at https://github.com/dalance/svls which comes from https://github.com/dalance/sv-parser which does very well at https://symbiflow.github.io/sv-tests