eirikpre/VSCode-SystemVerilog #41: Syntax highlighting can't keep up with Allman indentation style

gcrabbe, 2019-03-19:
The syntax highlighting feature has trouble with indentation styles, such as Allman, where the opening brackets are on a line of their own:

[image]

eirikpre, 2019-03-20:
Yeah, that it a limitation to the current implementation :(
It's part of the backlog to write the textmate matching more iterative, instead of matching per line like now.
The indexing features have been prioritized lately since they caused slowdowns.

eirikpre, 2021-01-19:
@jecassis You mentioned this issue in the PR as fixes. Does it do that on your end?

From what I can see textmate is only matching one line at a time. If it is going to match multiple lines, only the first word has to be in the "begin" expression.

jecassis, 2021-01-21:
I have been making additional changes, however it looks right on my side repository. I have a few more corner cases to address and I will submit another set.

[image]

eirikpre, 2021-01-27:
Should now be working correctly.
Big thanks to the great changes to the textmate setup by @jecassis

closed 2021-01-27 by eirikpre