eirikpre/VSCode-SystemVerilog #42: Incorrect highlighting for unique case and enumeration.

Xusine1131, 2019-03-23:
The syntax highlighting for unique case with two keywords in the same line is incorrect. It seems to be recognized as a module called unique instantiation:
[image]
And as you say, my enumeration values which are not capitalized can not be identified and highlighted but the capitalized one (EDONE) can.

eirikpre, 2019-05-16:
The matching of enumerated values are quite dumb, but lack of time and a good idea made it so.

It is just matching all uppercase words, and uses the same color as numbers.
Often uppercase words are either parameters or enumerated types.

Xusine1131, 2019-05-18:
Why not maintain a symbol list for enumerated value?

eirikpre, 2019-05-18:
The biggest problem is mainly that textmate (syntax coloring) is totally separate from the typescript scripts that parses the 'intelligently'.

Textmate afaik at least, doesn't have an easy way to store symbols encountered earlier in a file.

closed 2020-04-24 by eirikpre