eirikpre/VSCode-SystemVerilog #53: As of release 6.0. bracket matching for case statements don

bman844, 2019-07-03:


eirikpre, 2019-07-04:
Hm, that's strange, it's not showing like that on my end.
Perhaps you can see what you get out of:
[image]

If it's just my extension working on the document, hopefully it only matches with: _keyword.control.selection.systemverilog_ and _text.systemverilog_

bman844, 2019-07-16:
I did as you suggested, and noted that both keywords are identified with keyword.control.selection.systemverilog and text.systemverilog

Looking at the behavior without any other extensions installed in the most recent version, I noted an area it behaves strangely. 

[image] 

[image]

In the first image, the brackets appear to match when the cursor is on the the case keyword, while they no longer appear to match when the cursor is on the endcase keyword. Using the Ctrl+Shift+\ shortcut to jump to the matching bracket seems to show a similar story. The jump will jump for "case" to "endcase", but will not work for jumping from "endcase" to "case." This behavior does work fine with "begin"s and "end"s, but does not seem to handle "case"s and "endcase"s too well. I don't know if this is something you can replicate on your end.

bman844, 2019-07-16:
I decided to take the time to look through commits and see what the differences were. I think I found the potential issue and I imagine the fix might not be too bad, but I have no idea how to go about it since I have no experience personally with VS Code extensions. 

At a couple of points between version 5.6, and version 6.0, there were a couple of commits regarding adding support for randcase. in the language-configuartion. What I have noted is in the brackerts section of the language-configuration.json, there are two bracket types that end with endcase. You can see them here. Both case and randcase have keywords ending with endcase. I think that may be the reason as to why it may be able to jump forward, but not necessarily backwards. 

[image]

eirikpre, 2019-07-24:
Good find! Thanks for finding the source, I will see if it can match both case and randcase properly or change back to only case matching. Seems like what you found is me misusing the language config.
Will see to it next week after vacation :)

closed 2019-08-08 by eirikpre

bman844, 2019-08-09:
Glad to have helped! : )