eirikpre/VSCode-SystemVerilog #55: Go To Definition Inconsistent

carrola1, 2019-07-19:
Great extension! I've found the go to definition feature does not always work. I can jump to a module definition successfully, go back to the original file, try again, and it no longer works. After playing around a bit i found that if i save a file before using go to definition it works every time.

eirikpre, 2019-07-24:
Hm that is strange, big changes have been done to the backend storage of the symbols, namely also that they will update accordingly when a file is saved.
Will try to reproduce when vacation ends next week

eirikpre, 2019-09-05:
I have not been able to reproduce this error, is it still a problem?
Since then, I changed many of the types to built-in standard types, which may have solved your problem.

mahsoommoosa42, 2019-10-27:
I am facing this problem. Support for definitions is very erratic and it doesn't work most of the time. And I think the number of indexed objects is way less than there actually is. Btw, great extension. Real life-saver.

eirikpre, 2019-10-28:
Mhm, that is interesting, I did not find any way to figure out what are missing.
I would probably guess that you are not getting definitions for signals in the port list inside the module.
I'm in the process of adding that, but I'm kinda waiting for this Code issue: https://github.com/microsoft/vscode/issues/53034 
(because outline will explode when adding the ports to it)

eirikpre, 2019-10-28:
And the parser isn't that clever, it will not understand multiple declared variables on one line for example.
```
logic will, not, understand;
```
vs
```
logic will;
logic understand;
```

If you are familiar with regexes, this the matcher looking for instantiated objects: 
https://github.com/eirikpre/VSCode-SystemVerilog/blob/master/src/parser.ts#L52

mahsoommoosa42, 2019-10-28:
I just debugged the extension. this.symbolsCount returns 79 but, when i checked the contents of this.symbols. Only 22 objects were present within and only the names of the modules in my files. The signals within these modules were not present (even for my currently open file).

eirikpre, 2019-10-28:
Yeah, this.symbols would be a list of files containing modules. So you might have multiple modules per file, but that shouldn't be a problem

mahsoommoosa42, 2019-10-28:
Yes, but within them I believe only the modules are listed. Go to definition works on instantiated modules but not on any of the other objects.

Reopened #55<https://eur03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Feirikpre%2FVSCode-SystemVerilog%2Fissues%2F55&data=02%7C01%7C%7Ca0f29dea255f403a81b008d75bbdd459%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637078743176834091&sdata=7RDNtQdkVVRAYrHWENu9BqyMzTCw5fOlhdJCGiuk08A%3D&reserved=0>.

mahsoommoosa42, 2019-10-28:
I'll test with the new commit and see if it does the trick.

eirikpre, 2019-10-29:
It is primarily modules that will be indexed, but depending on the size of the workspace, it will index packages, classes, macros, ++ and 1 level of hierarchy (eg. class within package, functions within class).

I will try to rewrite the regex looking for symbols to include multiple declared signals, and then release on the marketplace.

mahsoommoosa42, 2019-10-29:
Okay. Didn't notice that the extension doesn't index signals. Let me know if you require assistance. More than happy to contribute.

eirikpre, 2019-11-01:
@mahsoommoosa42 I published the changes which should match a lot more symbols. Tell me if you if(when?) encounter any issues, or if there is something you are particularly interested in contributing to.

closed 2019-11-01 by eirikpre

mahsoommoosa42, 2019-11-06:
Working great on first look