eirikpre/VSCode-SystemVerilog #62: Compile command does nothing after updating to 0.8.5

AitBits, 2019-09-10:
When I tried to compile a file through the command palette or compile on save option, it does nothing and returns no output.

eirikpre, 2019-09-10:
Ah noes :( 

I'll look into it soon

closed 2019-09-10 by eirikpre

AitBits, 2019-09-11:
That commit just looks like a version bump to me, and the problem persists on my side. I think you forgot to stage some file?

eirikpre, 2019-09-11:
Ah yes, I made a typo in the commit message, so I linked the wrong issue.
Here is the commit: [62c235c0846e2db17fb000dd7385a859ec052136 ](https://github.com/eirikpre/VSCode-SystemVerilog/commit/62c235c0846e2db17fb000dd7385a859ec052136 )

That fixed it on my part, the output is piped to Output/System Verilog Language Server when the server logs.

[image]

AitBits, 2019-09-16:
Yeah, that seems got fixed but I'm getting another problem. I will open a separate issue.

commit 94c1405e (2019-09-10): 0.8.6, fixes #62