eirikpre/VSCode-SystemVerilog #63: Files are not getting passed to compile command

AitBits, 2019-09-16:
Previously, opened files got fed into compile command as input.
With current version, files don't get passed and compile command just run by itself resulting in an error.

`%Error: verilator: No Input Verilog file specified on command line, see verilator --help for more information`

eirikpre, 2019-09-16:
Ah damn sorry, I'm not using it actively, so it passed right by me. :(

I will look into it this week.

In the meantime you can use a build task if you want. See the second task in here:
[tasks.json](https://github.com/eirikpre/VSCode-SystemVerilog/blob/master/.vscode/tasks.json) 
Replace command=`wsl` with `shell` and remove `wslpath` in the arguments.

eirikpre, 2019-09-20:
Should be fixed with 0.8.7 :)

AitBits, 2019-09-23:
That was quick, thank you for the work. It runs perfectly now!

eirikpre, 2019-09-23:
No problem, just a silly mistake by me.

Thanks for reporting

rkrajnc, 2020-04-07:
Hi, this seems to still be a problem with current versions on Windows, I tried 0.9.5, 0.9.4 and the 0.8.7 version mentioned here, none of them attach the filename to the Verilator launch command.

rkrajnc, 2020-04-15:
Hi, thanks for reopening this. Is there anything I can help you with to get this fixed?

eirikpre, 2020-04-17:
You could try debug the extension, but I'm leaning toward people creating their own build tasks, instead of piping through the extension.
I have an example here:
https://github.com/eirikpre/VSCode-SystemVerilog/blob/master/.vscode/tasks.json

eirikpre, 2020-04-23:
Published a tiny patch that gives an completion suggestion in all `tasks.json`. Now you should be truly free to give all the arguments you want. Can also serve as a template to run other tools :)
[image]

Becomes:
[image]

joecrop, 2022-03-12:
Closing because of no comments in almost 2 years

closed 2022-03-12 by joecrop