eirikpre/VSCode-SystemVerilog #64: Index typedefs

Jibbajabbafic, 2019-09-27:
The extension doesn't seem to pick up on typedefs which would be very useful.

Example where uint_t isn't recognised:
```SV
class typedef_test;
    typedef int unsigned uint_t;
    uint_t number;
endclass
```

closed 2019-09-30 by eirikpre

Jibbajabbafic, 2019-09-30:
Amazing, thank you!

Jibbajabbafic, 2019-09-30:
@eirikpre There are still some typedefs which don't get picked up:

```SV
class typedef_test;
    typedef int unsigned uint_t;
    typedef logic [0:15] packed_array_t;

    typedef int unsigned uint_array[10];
    typedef int unsigned uint_queue[$];
    typedef int unsigned uint_dynamic_array[];
    typedef int unsigned uint_associative_array[uint_t];
    typedef int unsigned uint_array_2d[10][10];
    typedef logic unpacked_array_t [10];
    typedef logic [0:15] packed_unpacked_array_t [10];

    // Working:
    uint_t number;
    packed_array_t packed_array;

    // Not working:
    uint_array array;
    uint_queue queue;
    uint_dynamic_array dynamic_array;
    uint_associative_array associative_array;
    uint_array_2d array_2d;
    unpacked_array_t unpacked_array;
    packed_unpacked_array_t packed_unpacked_array;
endclass
```

eirikpre, 2019-10-02:
Ah what a fantastic class to test this on.
Next version should match all of the above correctly.
I'm not being thorough enough when working with these fixes..

Jibbajabbafic, 2019-10-08:
Awesome thanks, I can provide plenty more for testing if you need haha

Jibbajabbafic, 2019-10-08:
I've also noticed the cursor jumps to the beginning of the line instead of the type so you can't rapidly jump through definitions using F12 like normal variables.

commit 04533592 (2019-09-30): 0.8.9
fixes #64