eirikpre/VSCode-SystemVerilog #66: Allow include directories to be defined for code definitions external to the VS code workspace

paulmckechnie, 2019-11-07:
The plugin would be enhanced by allowing the paths to external code libraries to be specified. For example, the Microsoft C/C++ plugin allows include directories to be specified so that function/class definitions can be presented and followed. This could be useful to reference the UVM library source code for example as this is not normally included in the workspace itself. Many developers have other libraries that they wish to use but not modifiy in their projects and which might be external to the VS Code workspace.

eirikpre, 2019-11-08:
That's a really good idea!
I don't have the time to add this during November, but this has priority on next dev session.

eirikpre, 2020-04-24:
I added `includeIndexing` as a setting so you can configure which files are being indexed.
If an entry in the setting does not start with * it will treat the path as an absolute path.

Published the extension with this feature just now
Tell me if you encounter any issues! 

Limitation: The case where relative paths are used without wildcard in the start will be treated as absolute paths.

eirikpre, 2020-04-24:
There seems to be an issue with this version where it is not activating properly, debugging ongoing

closed 2020-04-24 by eirikpre