eirikpre/VSCode-SystemVerilog #67: UVM support?

davidpodless, 2019-11-10:
Hi, is it possible to recognize and auto-complete functions from UVM? if not, is there any future planning for that?
thank you

eirikpre, 2019-11-13:
Yes, I do think you will get one such feature in the future. 
It will not be a custom solution for uvm, but general code completion for all functions found in the design.
Related issue: #66

paulmckechnie, 2020-03-05:
One approach to including the UVM libraries for parsing is to add an additional folder to the VS code workspace. This works with VS code as is. 

That said, the additional folder in the workspace can get annoying especially if working over SSH.

eirikpre, 2020-05-06:
Heia, I think solutions for this exists:
The setting `includeindexing` can include additional folders to be indexed (thus providing lookup).

And I recall there being another extension with quite thorough snippet support for UVM:
https://github.com/gvekony/sv-1800-2012 

Please report if you have any issues with includeIndexing

closed 2020-05-06 by eirikpre