eirikpre/VSCode-SystemVerilog #69: Support parameters in "Go to Definition" or "Peek Definition"

vityokm, 2019-11-12:
Hi,

I have bunch of modules that are configured with parameters. Normally there are 2 files: module and parameters. The parameters are constant and there are a lot of them. So to see what a parameters equals to, I have to copy the name, go to the parameters file and search for it to see what it is. Which is a pain.

Selecting "Go to Definition" or "Peek Definition" shows a horizontal line scrolling at the top of the editor and nothing happens.

Thank you!

eirikpre, 2019-11-13:
I agree with you, that would be very useful.
Today, some parameters are parsered if they are described in the same file. I think this feature is can be easily created if the definition provider follows this kind of usage `package::parameter`, similar to how ports are read. 
I think there should be enough time to implement this before Christmas. :)

eirikpre, 2020-05-06:
Heia, I tried to do this before but it didn't quite work. I made a fix now that lets you investigate objects if it's prepended by the package.

If a package is imported with `import` it's unfortunately smart enough to look in that, it's not fully scope aware.

I'll update the marketplace shortly.

closed 2020-05-06 by eirikpre