eirikpre/VSCode-SystemVerilog #71: Could this plugin be used with coc.nvim?

antoinemadec, 2019-12-17:
I've been using your plugin in VSCode and I like a lot!
I was wondering if it would be doable to port it to **coc.nvim**?

**coc.nvim** supports LSP servers: https://github.com/neoclide/coc.nvim/wiki/Language-servers

Thanks in advance,
Antoine

eirikpre, 2019-12-18:
In the future, yes :)

Most of the features are currently provided by the client, using regular expressions to parse the code.

The idea in the future is to move everything over to the LSP with a more powerful parser, but current state of affairs is that it is in development: #68 

You can use the LSP now, but it wont provide much.

antoinemadec, 2019-12-18:
@eirikpre I will keep an eye on this :smiley: 
Could you please let me know how to compile the LSP server, just to give it a try?

For now, I will keep on using the universal Ctags, but this really is exiting news!