eirikpre/VSCode-SystemVerilog #74: Extension generates too many logs

Jibbajabbafic, 2020-01-14:
I'm using this extension with Remote SSH and after reloading the VSCode window a few times I get over 250mb of logs in ~/.vscode-server/data/! This is not ideal as I only have a limited amount of space for my home area on the host machine.

After indexing my workspace there are about 3500 files and it appears the contents of every single file is being sent to the logger by default which should probably only happen during debug.

I think this started happening after the recent update that added the ANTLR parsing feature (0.9?).

Contents of one of the folders:
```
~/.vscode-server/data/logs/20200114T154058/exthost3/output_logging_20200114T161336:

   30.1 MiB [##########]  5-System Verilog Language Server.1.log
   26.0 MiB [########  ]  5-System Verilog Language Server.log
   12.0 KiB [          ]  1-Git.log
    4.0 KiB [          ]  2-Git Graph.log
    0.0   B [          ]  4-SystemVerilog.log
    0.0   B [          ]  3-GitLens.log
```

### VSCode Info:
Version: 1.41.1 (user setup)
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-18T14:58:56.166Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 10.0.17763

### Relevant Extensions:
  - Name: SystemVerilog - Language Support
    Version: 0.9.2
  - Name: Remote - SSH
    Version: 0.48.0

eirikpre, 2020-01-14:
Oh shit that's a serious issue.
I'll look into it tomorrow

eirikpre, 2020-01-15:
Fixed with 0.9.3, should be arriving soon. Just published to the marketplace :)

closed 2020-01-15 by eirikpre