eirikpre/VSCode-SystemVerilog #75: LSP Server: Consider integrating svls into this extension

Jibbajabbafic, 2020-01-14:
I found another SV extension which uses a Rust based language server which looks quite promising, but the VSCode extension for it has not been developed as much as this one

Rust has the potential to be a lot more performant than TypeScript, and a lot of the groundwork has already been completed so it might be worth integrating it into this at some point.

svls repo:
https://github.com/dalance/svls

svls VSCode extension:
https://github.com/dalance/svls-vscode

Jibbajabbafic, 2020-01-15:
The SV linter is available separately too so only that part could be integrated if changing the language server is too much.

sv-lint:
https://github.com/dalance/svlint

eirikpre, 2020-01-15:
I'll have a look, but cannot promise anything, at the very least not in the close future.
Priority is to get a grasp on ANTLR so it can be used efficiently, then move some of the current features from the client to the server.

Jibbajabbafic, 2020-01-15:
Sounds good

philipaxer, 2022-01-05:
I think over long-term the verible tools are more promising compared to svls
https://github.com/chipsalliance/verible

Jibbajabbafic, 2022-01-05:
> I think over long-term the verible tools are more promising compared to svls
> https://github.com/chipsalliance/verible

Yes that also looks quite promising now so I've updated the issue to reflect this