eirikpre/VSCode-SystemVerilog #76: 'Go To Definition' Fails Across Files With Specific Multi-Line Comment

onesmallskipforman, 2020-01-18:
The 'Go To Definition' feature displays "No definition found for '<target_module>'" whenever target_module is in another file and target_module is preceded in the file by a multi-line comment that meets all of the following criteria:

- The comment contains the text 'module' (all lowercase, standalone word).
- The text 'module' has either whitespace to the left or no text at all.
- The text 'module' is followed by any alpha-numeric text, either on the same line or a line below, with no other non-whitespace characters in between.

Examples of comments that cause the error:

    /*
    module abc
    */

     /*
                   module 123
     */

    /* dog

          module
    cat */

Examples of comments that do not cause the error:

    /*module abc
    */

    /*
    moduleabc
    */

    /*
   
                text module 123
    */

    /*
    module (  123
    */

    /*
    Module www
    */

    /* dog
         module
    */

eirikpre, 2020-01-24:
Interesting case, will have a look into it when there is some time :)

Good job on describing the issue including reproduce cases!

eirikpre, 2020-04-24:
I think this is fixed now with the updates to the definitionProvider, I did not encounter any issues while testing with mentioned commit.  Reopen if the issue still persist.

closed 2020-04-24 by eirikpre