eirikpre/VSCode-SystemVerilog #78: Add array support for Exclude Indexing setting

Jibbajabbafic, 2020-02-06:
It's not clear right now if multiple exclude indexing globs are supported so it would be nice to implement the setting as an array like the default File Exclude VSCode setting.

Example:
[image]

How it appears in the settings:
```
		"files.exclude": {
			"**/.git": true,
			"**/.svn": true,
			"**/.hg": true,
			"**/CVS": true,
			"**/.DS_Store": true
		}
```

eirikpre, 2020-02-10:
I think it is supported through glob with {<a>,<b>}.

I can make the setting an array and concat, but low prio at the moment

Jibbajabbafic, 2020-02-10:
Alright thanks, I'll try that method

eirikpre, 2020-04-24:
Doing a bit of spring cleaning, if the fix mentioned did not work, reopen the issue

closed 2020-04-24 by eirikpre