eirikpre/VSCode-SystemVerilog #79: Support Predifined Variables for "launchConfiguration"

Xusine1131, 2020-03-03:
https://code.visualstudio.com/docs/editor/variables-reference

At present, the lint command is  "launchConfiguration" + ${file}, and any VSCode predefined variables are not supported. However, it's quite normal to preprocess the opening file before pass it to the linter.

eirikpre, 2020-03-17:
Yep, preprocessing is not implemented yet. Work on-going there, but it will take some time before I can make it.

eirikpre, 2020-05-07:
Hey @Xusine1131 
Have you tried using a task? (Tasks can now be defined for a user, instead of per workspace)

Snippet here:
https://github.com/eirikpre/VSCode-SystemVerilog/issues/63#issuecomment-618331523

Xusine1131, 2020-05-07:
Thank you @eirikpre 
It's nice to regard linting as a task. But I think it's not very convenient since I let the linter automatically run when saving files. Is it possible to automatically run a task on saving the file?

eirikpre, 2020-05-07:
Yes, I'm certain I've seen an extension that runs a task on save :)
I haven't tried it myself, but maybe this will do the trick? https://marketplace.visualstudio.com/items?itemName=emeraldwalk.RunOnSave

Xusine1131, 2020-05-07:
Thank you @eirikpre !
I think you can refer to that extension since it can run commands with predicated Variables. If the parsing function is found, this issue can be directly solved!

Xusine1131, 2020-05-07:
https://github.com/emeraldwalk/vscode-runonsave/blob/master/src/extension.ts#L195
I think what he did is just replacement.

eirikpre, 2020-06-22:
Just keeping the task open to formally describe this on the wiki and remove the internal launch of linters :)

closed 2021-01-18 by eirikpre