eirikpre/VSCode-SystemVerilog #80: Non-matching regex in variable.

Obliged, 2020-03-16:
As far as I can tell the regex `(?=\.)\w+` does not match anything. It looks for `.` within a series of word characters which does not include `.`.

https://github.com/eirikpre/VSCode-SystemVerilog/blob/62246528a95f1ff2c71cecad1be4dcdfec335478/syntaxes/systemverilog.tmLanguage.json#L353

eirikpre, 2020-03-17:
Hm, I think my intent was to catch hierarchical references to objects. Typically when you force or probe a variable in the design you write it like this: `dut.module.signal`

Obliged, 2020-03-17:
Wouldn't that just be `\w+\.`  ?

I am probably misunderstanding how these regexes are applied, because I don't understand how the match above works either:

https://github.com/eirikpre/VSCode-SystemVerilog/blob/62246528a95f1ff2c71cecad1be4dcdfec335478/syntaxes/systemverilog.tmLanguage.json#L349

Which I (and regex101.com) read as  look for whitespace or `.`, but not \`  within more than one word character (which does not include whitespace. `.` nor \`) and the final lookahead i don't understand. Where is it looking into?

eirikpre, 2020-04-24:
Heya, sorry for not following this up recently.
The textmate grammar is applied multiple times per line, and may work a bit strange sometimes. I agree that these regular expressions doesn't seem right. I think what I did at the time was to not match variables because it is so hard to cover all cases. Eg:
[image]

There might be possible to provide additional syntax highlighting through smart parsing, but that is for the future.
I tried fixing the regex in question, and it did only make things uglier 😄 
Removing the whole variables section is better, `this.<member>` is still matched with keywords.

Thanks for the ticket, sorry for using so long time to answer!

closed 2020-04-24 by eirikpre

Obliged, 2020-04-24:
Sounds like a good solution. I only found this because I was using your code as a baseline for creating a VHDL-grammar. So the issue was never visible for the user, I was just confused as to what I should translate it into ^_^'