eirikpre/VSCode-SystemVerilog #83: Indexing Always on Open

Alex-Mann, 2020-04-04:
I am using VSCode remote ssh extension to edit some files. Each time I connect, the extension reindexes and just hammers my CPU. Is there a way to cache results so that it doesn't reindex everything each time I connect, but rather does things incrementally?

eirikpre, 2020-04-05:
Unfortunately, I do not have a nice way to store the data in-between opens.

But you can turn off auto-indexing on startup with the `disableIndexing` setting. Then if you want indexing, click on SystemVerilog on the status bar.

Alex-Mann, 2020-04-08:
I have no idea what the underlying implementation looks like.. but is there any possibility of storing a cache like data structure in a file and periodically updating it so that the indexing doesn't start from square one? Or is the current implementation not really conducive to something like this?

eirikpre, 2020-04-09:
Think I figured it out, I published a patch.
Can you test it for me? :)
Ps turn off disableIndexing

Alex-Mann, 2020-04-09:
Sure, how do I pull the changes of the patch into the extension? Or have you already pushed?

eirikpre, 2020-04-09:
It should update shortly on the marketplace, it is usually up in less than 10 min. Then you just need to reload. You should be able to check the version in the extension sidebar, new one is 0.9.7

Alex-Mann, 2020-04-09:
Alright so I've updated everything, but if i disable indexing I no longer get the cmd+click to view other modules right?

eirikpre, 2020-04-10:
If the object you're looking up is in the same file, it should find it.
Also, it does incremental indexing when you open verilog files.

Right now it stores the state after full blown indexation, it's probably better to store it on workspaceClose to get the incremental objects too..

eirikpre, 2020-04-10:
After some trying out, I think it works. Added saving the indexed objects on incremental updates too.

closed 2020-04-10 by eirikpre

Alex-Mann, 2020-04-15:
So should I keep indexing disabled or can I now turn it back on? I'm still a little unclear on what the setting should be.

eirikpre, 2020-04-17:
That is up to you.
If you keep indexing disabled, indexing will not trigger automatically when you open previously un-indexed folders/workspaces. However, it will use the saved data from already indexed workspaces regardless of that setting.

If you would like to always control indexing by yourself, i would suggest keeping it disabled and click the statusbar when you would like to index (and that database will carry over next time you open that workspace/folder)

Alex-Mann, 2020-04-17:
I see. But in the event that indexing is enabled, is there a check to determine what to re-index (so that you don't reindex files that have not changed since the last index)? This would suffice so that on start up it doesn't always reindex everything over again if it hasn't changed.

eirikpre, 2020-04-18:
This feature will make it so that if it finds the cache from last time you opened the folder/workspace, it will not run indexing. (but instead use the cache) :)