eirikpre/VSCode-SystemVerilog #87: indexing is not working at version 0.10.4

wl6g11, 2020-04-27:
When I update to the version 0.10.4, indexing is not working any more, zero file is indexed after execute the index command.
Version 0.10.0 which I roll back to seems working OK.

eirikpre, 2020-04-28:
Can you check if you are using the setting `systemverilog.includeIndexing`?
That will include files by glob, don't know to debug since it seems to be working on my 3 computers

pavii, 2020-04-28:
Hey, I think I have a similar problem that happened around the upgrade to 0.10.0. I also updated verilator. I can see that verilator does the job in the output tab under System Verilog Language Server but it is not shown in the editor.

I am using Windows machine and everything is up to date

eirikpre, 2020-04-29:
Heia, a person pull requested in new matching algorithms for verilator because there is a new verilator version with new output formatting, maybe that's the issue? See #84 

Also, I added a snippet so you can tweak and launch verilator as you'd like, see #63 

@pavii Do you also have problems with indexing? (Those two are two separate systems)

wl6g11, 2020-04-29:
I tested on another computer, it works all right at version 0.10.4.
Well, it is very interesting, I will try to find out what is wrong.
Any way, thanks for your replay and I think this issue can be closed.

wl6g11, 2020-04-29:
Well, I figure out the problem.
The version 0.10.4 seems not indexing *.v file, it only indexs *.sv file.
That's why zero file is indexed in my code folder which only contains verilog file.

wl6g11, 2020-04-29:
Finally, the problem is the default seeting of systemverilog.includeIndexing ( **"**/*.{sv, v, svh, vh}"** ).
The problem is the **space character**, the *.svh and *.vh files will not be indexed as well.
I think the indexing code read the **space character**, then the ***.  v**file will be indexed, but not ***.v**file.
It works all right after I changed the seeting to "**/*.{sv,v,svh,vh}"

eirikpre, 2020-04-29:
Ah good catch! I'll have to fix this default setting pronto. Thanks for figuring it out

closed 2020-04-29 by eirikpre