eirikpre/VSCode-SystemVerilog #88: TypeError: Cannot read property 'uri' of undefined

crs-ica, 2020-05-16:
I am running VS Code 1.45.1.
When hoovering the code (.sv, .v), the following error is printed to console:

```
[2020-05-16 12:29:16.212] [exthost] [error] [eirikpre.systemverilog] provider FAILED
[2020-05-16 12:29:16.212] [exthost] [error] TypeError: Cannot read property 'uri' of undefined
	at c:\Users\turlea\.vscode-insiders\extensions\eirikpre.systemverilog-0.10.6\out\providers\HoverProvider.js:13:67
	at processTicksAndRejections (internal/process/task_queues.js:85:5)
```

[image]

eirikpre, 2020-05-21:
Ports aren't found, but it should reject, not throw a TypeError. Will look into it, thanks for reporting

eirikpre, 2020-05-21:
I think I got a fix, will release shortly.

commit c142cd68 (2020-05-21): Adding additional failsaves to hover and defproviders, fixes #88
closed 2020-05-21 by eirikpre