eirikpre/VSCode-SystemVerilog #89: Add support for "go to def" for const/enum/etc

dtalaski, 2020-06-04:
Hi Eirik,
Could you add "go to def" support for const, enum, localparam, parameter, struct, union types?

For example,
In a package file: (located in a parallel dir to the RTL files)

...
typedef struct packed
{
               logic hey;
               logic there;
} my_struct_s;
...

In an RTL file:

...
my_struct_s some_sig;

...

assign some_sig = ‘{hey: 1’b0, there: 1’b1};   // found on line 34

...

I would like to be able to hover (or ctrl-click) over “some_sig” on line 34 in the RTL file and see the definition of the struct.

dtalaski, 2020-06-05:
Digging into this a bit more, I found that "go to def" works somewhat.
If the definition is in the same file; enums, parameters, work
If the definition is in the same fille; const, localparams, structs, unions, do not appear work
If the definition is in a package: all types do not appear to work

arnonsha, 2020-06-08:
Hi
Yes, this is really annoying, especially when working with packages, where you have to jum between back and forth between actual RTL files, and the package files to better understand typedefs.
Seeing the correct definition (read from the pkg file) Would be a great help and very much appreciated.

thanks !

eirikpre, 2020-06-17:
Heya, yeah I'm seeing the same sometimes.
I think the problems have multiple causes:
1. If a variable/param/.. is declared and assigned at the same time it will not be picked up by the parser. Eg `localparam int a = 1;`
2a. `import package::*;` is not taken into consideration, nor is `include file.v`.
2b. I've tried to make the GoToDef smart enough to pursue the package if it detects the variable is prepended with a `<package>::<var>`, but since 1) is usually used in packages it won't find anything.

Should probably try fix 1), but it will cost some extra processing power as the regular expression is growing :)

denisrheault, 2020-12-17:
Hi Eirik,
Congrats for this wonderful extension.
Regarding 2 (import package::*;), I don't see it being a problem.
In the package, something like
  typedef logic [NUM_PORTS-1:0] data_t;
works fine from any file having 'import package::*;' but this doesn't work:
  typedef struct packed { logic [NUM_PORTS-1:0] data_t; logic rwn; logic en; } host_bus_t;
and neither does
  typedef struct { logic [NUM_PORTS-1:0] data_t; logic rwn; logic en; } host_bus_t;

We use typedef a lot and I constantly have to have a look.  It'd be great if that could be fixed.
Thanks!

jrhyde1, 2021-01-19:
has any progress been made on this issue? specifically, supporting structs in GoToDef and mouseover definition? This seems like a pretty high priority problem

For example: 

```
typedef struct packed {
    logic foo;
    logic bar;
} my_dumb_struct;

my_dumb_struct my_opp_struct;
```

Go to definition does not work for `my_dumb_struct`, ie, I can't see that it is made of `logic foo` and `logic bar`.

eirikpre, 2021-01-21:
I got the typedef structs working better now!
There is one noticeable inconsistency with the hover which I cannot explain.

This works:
[image]

Inconsistency: ( "{" is on the next line, jumping to it and peek still works fine)
[image]

It is not able to pick up directly instantiated structs:
[image]

eirikpre, 2021-01-21:
I will make a hotfix release which should help you navigate to typedefs; explained above.

Follow-up of the `include/import` discussion in #111.

closed 2021-01-21 by eirikpre

jrhyde1, 2021-01-21:
Awesome! and  #111 is much appreciated as well!