eirikpre/VSCode-SystemVerilog #91: Hiding regs/wires declarations from Outline panel?

ldm1417, 2020-06-16:
Hello,

Is it possible to hide/filter the regs/wires declarations from the Outline Panel? 

How can I leave just instance names in the Outline panel? 

Is there a way to add comments in the code, which would also appear in the Outline Panel? 

Thank you!

eirikpre, 2020-06-17:
Code has settings to hide things in the outline, I have this in my settings:
```
    "[systemverilog]": {
        "outline.showArrays": false,
        "outline.showVariables": false
    }
```
I have this image as a mapping guide: https://github.com/eirikpre/VSCode-SystemVerilog/blob/master/docs/SymbolKind_icons.png 

I don't think I will parse the comments, but full parse does look for labels, eg:
```
begin : my_label
    ...some code
end
```
then my_label should show in the outline

closed 2020-06-17 by eirikpre

ldm1417, 2020-06-19:
Could you please add a feature, which would also parse some regex in comments?
E.g. any comments, which start with **// -** will also parsed?
`// - THIS COMMENT WILL APPEAR IN OUTLINE`
This feature could be very useful in the Code Navigation!!!

ldm1417, 2020-06-19:
Now according to the OUTLINE... 

It seems that the module ports and instance names are both parsed as "Fields". But I'd like having just the instance names in the OUTLINE without the long list of the module ports. How could I do that?

[image]

eirikpre, 2020-06-22:
I will not add parsing of comments, sorry. There should be a tool doing this for all files to be honest.

If you remove the comment on line 5 it will show correctly.
[image]