eirikpre/VSCode-SystemVerilog #92: hovering mouse over an instance port doesn't show port declaration

ldm1417, 2020-06-17:
Hello,

Usually, while hovering a mouse over a port of the module instance, a tooltip pops up and shows a declaration of the port (as it's defined in the module itself). 

Usually, this works but not always... 

I'm attaching two files - teng_top.v and eth_mac_wrap.v The teng_top hierarchy includes an instance of the eth_mac_wrap module. While hovering over the ports of the eth_mac_wrap instances, the pop-up tooltip says "No definition found for <port_name>". 

Is this a bug? Could it be fixed, please? All the mentioned files are attached. I've replaced the .v extension of the files with the .txt extension - just for being able to attach the files. 

Thank you!
[eth_mac_wrap.txt](https://github.com/eirikpre/VSCode-SystemVerilog/files/4793432/eth_mac_wrap.txt)
[teng_top.txt](https://github.com/eirikpre/VSCode-SystemVerilog/files/4793433/teng_top.txt)

ldm1417, 2020-06-19:
Any response?

eirikpre, 2020-06-22:
I also find it kinda strange, this is the regular expression looking for ports:
```(?<!^(?:\/\/|`|\n).*?)(?<=\b(input|output|inout)\b\s*(?<type>(?:`?\w+)?\s*(\[.*?\])*?)?\s*(\b\w+\s*,\s*)*?)(\w+)(?=\s*((\[.*?\]\s*)*?|\/\/[^\n]*\s*)(?:,|\)))```

When searching in eth_mac_wrap it seems to find everything, but the parser doesn't pick up link_locked...

eirikpre, 2020-06-22:
Update: Hm, it is actually the comment on line 5, where it breaks up the module and the ports are perceived as signals instead of ports. 
I'm not able to fix this in the current parser due to preprocessing comments will mess with the character positions.

ldm1417, 2020-06-25:
Good job, Eirik! You actually found the problem! Once I removed the //* on the line#5 in the eth_mac_wrap module, this solved the problem! Now the ports of the eth_mac_wrap instance in the teng_top module are recognized while hovering a mouse over them. Well done, Eirik! Please close the issue :-)

ldm1417, 2020-06-25:
BTW, do you parse the references and drivers of the signals in your extension? While hovering the mouse over the signals, only declarations are shown in the pop-up tooltip... 
Should some settings be changed for seeing the signal's drives and references?

eirikpre, 2020-06-25:
No, sorry, references and drivers are not parsed.

Good that it fixed the problem, it's a bit of a bummer that the parsing is sensitive to comments.

closed 2020-06-25 by eirikpre