eirikpre/VSCode-SystemVerilog #94: Excluding files from indexing - doesn't work

ldm1417, 2020-06-20:
Hi there,

 I'm trying to remove some files from indexing. I've added the following into the settings.json:

`"systemverilog.excludeIndexing": "**/*_inst.{sv,v,svh,vh}",` - so, I'd like to exclude all the files, which have the '_inst' string in their names. 

But the above setting did not work for me. Why? 

BTW, how is it possible to clear/reset all the indexes, which was previously done? 

Thank you!

eirikpre, 2020-06-22:
Hm there might be that it indexes those files if you open them through the incremental indexing feature.
Didn't think of that actually.

The example you provided above seems to exclude the files correctly. (however, it is added when opening the file)

eirikpre, 2020-06-22:
Published the fix referenced in v0.10.10

closed 2020-06-22 by eirikpre