eirikpre/VSCode-SystemVerilog #97: UVM extension

RomanLeonB, 2020-11-01:
Hi,
thanks for the great extension :).
Do you have plans about an UVM extension (maybe for the 20 important classes)?

Roman

eirikpre, 2020-11-03:
Heia!
Not planned for this extension. There is this extension catered towards UVM: [SystemVerilog-1800-2012](https://marketplace.visualstudio.com/items?itemName=gvekony.systemverilog-1800-2012 )
You can use the setting includeIndexing to include the uvm files which might work for you. (Most people do not have uvm in their workspace), previous comment https://github.com/eirikpre/VSCode-SystemVerilog/issues/67#issuecomment-624608387

RomanLeonB, 2020-11-04:
> Heia!
> Not planned for this extension. There is this extension catered towards UVM: [SystemVerilog-1800-2012](https://marketplace.visualstudio.com/items?itemName=gvekony.systemverilog-1800-2012 )
> You [...]

Ok thanks, is it possible to support the repository?

eirikpre, 2020-11-04:
Not sure what you mean by support the repository? :)

closed 2021-01-15 by eirikpre