eirikpre/VSCode-SystemVerilog #113: Syntax Highlighting

sconwayaus, 2021-02-01:
Hi,

Updated to version 11 and noticed a few odd things with the syntax highlighting. I've screen grabbed version 0.11.2 (latest as of today on the right) and version 0.10.11 (left).

[image]

Case statement shows different colors of STATE_1 an STATE_2 on line 32.

FWIW, I also preferred the light green used for the parameter and names in the enumeration (lines 2, 3 and 12-16).

As an aside, the new highlighter didn't pick up that I typed "parameter" incorrectly on lines 2 and 3, where as the old syntax highlighting did.

jecassis, 2021-02-01:
Thank you for the example. Found a couple of other cases that needed correcting and will push an update. Specifically from the example line 32 has already been fixed by the `port-net-parameter` grammar. Also note that `typedef enum unsigned int` is not legal according to my reading of the LRM. The `unsigned` keyword should go last in this case.

As for the color preference, all the syntax highlighting does is classify each tokens to a specific scope. The colors are determined by your theme. You can override the theme's colors using `editor.tokenColorCustomizations` in `settings.json` to suit your tastes. For example the `STATE_*` parameters you mention are classified as `constant.other.net.systemverilog`. Note that this does not include any semantic information. To find the scope for each token run `editor.action.inspectTMScopes`.

The example from this issue using a minimally-modified One Dark Pro theme is shown below:

[image]

sconwayaus, 2021-02-02:
Thanks Jecassis,

You're right, enum should be "int unsigned" and thanks for the tip on the
theme settings. Looking forward to the update.

Cheers
Steve

bitrocco, 2021-02-04:
There is a related syntax highlighting issue.

When the definition of `enum` type is written on one line, the syntax highlighting is wrong, but it is correct in lowercase, as shown below:

[image]

jecassis, 2021-02-04:
Thanks @sconwayaus. @bitrocco, single-line typedefs like this are fixed in #112. That said, let me clarify what you should expect. There is a small concession made in the syntax matcher to mark identifiers in all caps using the constant classification/coloring. Without semantic information, the matcher prioritizes this as a best guess based on the various coding styles I have come across in the industry such as [this one](https://github.com/lowRISC/style-guides/blob/master/VerilogCodingStyle.md). Many lint tools even enforce these kind of conventions in their [default ruleset](https://google.github.io/verible/lint.html#parameter-name-style).

[image]

bitrocco, 2021-02-04:
@jecassis Thanks for your reply.

Sorry for the previous unclear explanation, but the original intention is that the enum_identifier in the enumeration declaration should have the same highlight color. However, the #112 solution is also accepted, because it generally follows the same code style. So look forward to the next version update :smile:.

eirikpre, 2021-06-04:
Should be ok with the last release 0.11.3 :)

closed 2021-06-04 by eirikpre