eirikpre/VSCode-SystemVerilog #115: verible-verilog-format doesn't work

xanthosvlachos, 2021-02-26:
Hello,

I have installed verible-verilog-format in linux by downloading the binaries for my distro. Added them to the $PATH, too.
From command line when i run the verible-verilog-format I see the correct formated file in the stdout.
When I run this from the vscode (Format using the System Verilog Language Support) , I see nothing changing in my file format.
Settings are untouched and I see that the custom command is set to verible-verilog-format as it was the default value there.

Can somebody help?
Thank you in advance.

eirikpre, 2021-04-19:
I've been quite busy at the moment and haven't had the time to toy with the feature too much yet, but I think @jecassis might be able to help :)

jecassis, 2021-04-19:
If the tool is already in `PATH`, ensure the `systemverilog.formatCommand` option is: `verible-verilog-format`

You can add any options to the command to customize the formatting. Using `Shift`+`Alt`+`F` or the `Format Document` command from the palette should run Verible (if there are no syntax errors or unsupported Verible constructs).

tolisaverkios, 2021-05-12:
Hi @xanthosvlachos.

I have exactly the same issue. verible-verilog-format is in my PATH, I can use it from my terminal, but VScode says that it can't find the verible-verilog-format command when I format the document with `Shift+Alt+F` or `Format Document`

FYI I managed to make it work by providing the full path of the executable.
So if the exe is under `/home/userX/bin/verible-verilog-format`, then either update the` Sytemverilog: Format command` in the settings with `/home/userX/bin/verible-verilog-format`, or add the following to the JSON of the settings: `"systemverilog.formatCommand": " /home/userX/bin/verible-verilog-format"`

eirikpre, 2021-05-12:
> I can use it from my terminal, but VScode says that it can't find the verible-verilog-format command
I think this might be because VSCode usually utilize non-login/interactive shells, and most people only change the path in their .*rc files, meaning it only applies to interactive shell. Also, changing path in the interactive terminal doesn't change path for the parent (VSCode itself) process.

jecassis, 2021-08-30:
You can download the `verible` utilities (or build your own) from this [repository](https://github.com/chipsalliance/verible). You'll want to place the utilities in your `PATH` or give `VSCode-SystemVerilog` a full path.

The style can be slightly changed in `verible` through the command-line but it is not crazy flexible; as an example, these are my settings in VS Code:

```json
"systemverilog.formatCommand": "verible-verilog-format --assignment_statement_alignment=preserve --case_items_alignment=infer --class_member_variables_alignment=infer --formal_parameters_alignment=preserve --formal_parameters_indentation=indent --named_parameter_alignment=flush-left --named_parameter_indentation=indent --named_port_alignment=flush-left --named_port_indentation=indent --net_variable_alignment=preserve --port_declarations_alignment=preserve --port_declarations_indentation=indent",
```

ezio1996, 2021-08-30:
> You can download the `verible` utilities (or build your own) from this [repository](https://github.com/chipsalliance/verible). You'll want to place the utilities in your `PATH` or give [...]
thanks so much! i have download the file, and put it in windows, and set the vscode systemverilog language support extention setting：  format command : "I:/verible-master/verilog/tools/formatter/git-verible-verilog-format.sh"
[image]

 and i opened a .sv file to see how it work,but actually there is nothing happened, could you please give me a guide in detail?
thanks again!!!

jecassis, 2021-09-01:
To close this conversation out Verible doesn't work on Windows (at least there is no official binary release for it). There are only releases for CentOS and Ubuntu. Any other distribution or OS, you'll have to build the repository yourself to get executables that you would point VSCode-SystemVerilog to call on your behalf. If you're very savvy you might find a way to use WSL but that is way beyond the support we can provide here as this is not the proper forum for that.

closed 2022-03-09 by joecrop

TCWORLD, 2022-05-31:
Just FYI, there are now Windows binaries available from the Verible releases page: 
https://github.com/chipsalliance/verible/releases

They have to be manually installed somewhere (just extract the ZIP, and dump the files somewhere useful), and the folder you saved the exe's to added to the PATH environment variable.

Once done the document formatting command works perfectly on Windows.