eirikpre/VSCode-SystemVerilog #125: Syntax highlighting errors

sconwayaus, 2021-06-22:
Hello,

Found a couple of syntax highlighting corner cases.

1. Declaring a class variable (maybe with parameters) causes the next control statement to highlight incorrectly (in this case the "if" on line 18)
[image]

2. When instantiating a module, adding a comment after the parameter list causes the instance name to highlight incorrectly (line 46).
[image]

sconwayaus, 2021-06-22:
Also Line 5, keyword "function" is not highlighted correctly.
https://user-images.githubusercontent.com/6986019/123011387-97b2cd00-d403-11eb-82fe-401f42c00eda.png

sconwayaus, 2021-07-05:
Another corner case when importing DPI-C function.

[image]

joecrop, 2022-03-10:
@sconwayaus I am able to reproduce (and solve) issues 1, 3, and 4. But my test case for issue 2 dies not fail when I add a comment on any line. Can you re-confirm issue 3?

For issue 1 and 3. The syntax highlighting was actually wrong. The symbol should have been identified as `storage.type.user-defined.systemverilog` which is a dark blue color like this:

[image]

I kind of like the green, but the blue is technically correct. Do you have any concerns with it being blue moving forward?

I have not looked at issue 5 yet.