eirikpre/VSCode-SystemVerilog #131: Go To Definition failed for specific signal name

TPLeung, 2021-09-22:
Hello,

I find it strange in my verilog code that the "go to definition" feature can work for some signal names but not for others. After I dig this problem deeper, I find signal names beginning with "if/else/for" will crash the feature. For example, "ifm_data" and "ifm_valid" in my case.

Also I exam the signal name beginning with other control words like "while/case/always", but those words will not crash the feature.

Could it be fixed?

Thank you!

TPLeung, 2021-09-23:
Updates: here add some example code snippets

Signal names with some control words like 'if | else | for | generate' cannot 'go to definition':

[image]

[image]

Signal names with other control words like 'while | case | default' can 'go to definition':

[image]

[image]

Also I've found the control words crashing the feature lies in the source code in https://github.com/eirikpre/VSCode-SystemVerilog/blob/bbfce884dc0e07f64e26c6d3c66ad5272e8aa241/src/parser.ts#L5. I am not familiar with Typescript so I don't know if this line of code causes the problem.

Anybody can help?

joecrop, 2021-11-01:
I can help... But it seems like @eirikpre has gone dark for the past several months, so it will be unlikely that we will get a fix merged. @eirikpre please let me know if it will be worth my time to create an MR for this.

joecrop, 2022-01-26:
@TPLeung I have fixed the problem and created #141 to be merged, If you have time, please pull and give it a try, or provide feedback at the least. Otherwise we can wait for the next release of the plugin.

TPLeung, 2022-01-27:
> @TPLeung I have fixed the problem and created #141 to be merged, If you have time, please pull and give it a try, or provide feedback at the least. Otherwise we can wait for the next release of the plugin.

Thanks for the reply and the contribution! I try to pull the codes and apply the newest commits but was stuck into several npm errors during the build process. Anyway, the modified codes seem to work on my problem and I will wait for the next release.

PR #141 (2022-02-10): added word boundaries around illegal symbols
Closes #131
closed 2022-02-10 by joecrop