eirikpre/VSCode-SystemVerilog #137: Allow selecting your syntax highlighting if 2 extension offer syntax highlighting for SystemVerilog

mrvkino, 2021-12-02:
Hi,

I am using two extension for my SystemVerilog:
[Yours](https://github.com/eirikpre/VSCode-SystemVerilog)
[Systemverilog-1800-2012](https://github.com/gvekony/sv-1800-2012)

The reason is they have a lot of snippet for UVM and he updates the list fairly regularly. But activating both of them, causes their syntax highlighting to replace yours (which is more to my liking).

I know that extension can offer syntax highlighting for the same language and you can select them.
[image]

Do you think it could be possible to do the same for your extension?
[image]

In the mean time I can copy their snippet, but it gets tedious after a while.

Thanks

joecrop, 2022-03-09:
Hi @mrvkino, where are you getting those dropdowns. I can't find them. I'm happy to enable this, but I can't find where it would be done.

joecrop, 2022-03-10:
OK, I think I understand how this works. The reason the log extension works is because it uses a different `language id` from the default vscode handler (I think). It seems like we don't really have that option here. If I change the `language id` from the vscode standard of `systemverilog` to something else, the tool will show up as a different option on the list. But, because the `language id` is different than what vscode recognizes the file to be, it doesn't load the extension on startup. It seems like a non-starter to require the user to always manually enable the extension every time vscode is launched, or even the first time frankly.

I think  the best option here would be to convince the maintainers of [Systemverilog-1800-2012](https://github.com/gvekony/sv-1800-2012) to collaborate, at least on sharing their snippets.