eirikpre/VSCode-SystemVerilog #138: The input and output signal cannot jump to the definition

guojiazhuxi, 2021-12-12:
[image]
The signal defined above cannot be jumped to the definition and cannot hover the mouse over the variable declaration.

joecrop, 2022-01-26:
@guojiazhuxi are you referring to the parameters, or the inputs? The inputs are already the definitions, so there is nowhere to jump to, right?

As for the parameters, are you hoping that the tool will follow the module instantiations up in the hierarchy?  In the case where a module is instantiated twice, there is no way to know which instantiation to trace the parameters to. After the 'reference provider' feature is released, you will be able to find usages of modules within your workspace, so that will help a lot with this problem.

guojiazhuxi, 2022-01-27:
> @guojiazhuxi are you referring to the parameters, or the inputs? The inputs are already the definitions, so there is nowhere to jump to, right?
>
> As for the parameters, are you hoping that the tool [...]

As shown in the figure above, the input / output signal is defined. When I see this signal in the code, I can't give the definition of this signal by placing the mouse on this signal, and I can't jump to the defined position of this input / output signal by using Ctrl

joecrop, 2022-01-27:
@guojiazhuxi There are two ways you can interact with the code:

1. **Go To Definition**: This is where you ctrl-click on a signal name somewhere in the file and it takes you to the definition. i.e. you ctrl-click on:
```
assign my_signal = 1'b0;
       ^
```
And it takes you to:
```
input my_signal,
      ^
```
This should work well 99% of the time. If this is not working, we need a specific example of when this does work, so we can fix it.

2. **Find References**: This is effectively the opposite of 'Go To Definition' where you ctrl-click on the definition, and it gives you a list of all the references of that definition in your code. i.e. you ctrl-click on:
```
input my_signal,
      ^
```
And it gives you a list:
```
assign my_signal = 1'b0;
       ^
assign another_signal = my_signal;
                        ^
```

Find References has been implemented bot note released yet.

guojiazhuxi, 2022-01-27:
[image]
If this is a variable defined internally, when I put the mouse it on it, it will automatically display the definition, and can jump to the defined position via Ctrl, as shown in the figure above.
However, if it is an input / output signal, I put the mouse over it, it will not display any, and pressing Ctrl will not jump, as shown in the figure below
[image]

guojiazhuxi, 2022-01-27:
> @guojiazhuxi There are two ways you can interact with the code:
>
> 1. **Go To Definition**: This is where you ctrl-click on a signal name somewhere in the file and it takes you to the definition. i.e. [...]
@joecrop This file uses the system Verilog language

guojiazhuxi, 2022-01-27:
[image]
If I press F12, it will show that no definition of it has been found， as shown in the above figure.

guojiazhuxi, 2022-01-27:
> 

If I press F12, it will show that no definition of it has been found， as shown in the above figure.

[image]

joecrop, 2022-01-28:
Thanks @guojiazhuxi, this is really helpful. Now I fully understand the problem. Let me look into this in the coming days.

guojiazhuxi, 2022-02-07:
> Thanks @guojiazhuxi, this is really helpful. Now I fully understand the problem. Let me look into this in the coming days.

thank you!

joecrop, 2022-02-18:
@guojiazhuxi, Can you please try the following settings:
- `systemverilog.documentSymbolsPrecision = "full"`
- `systemverilog.forceFastIndexing = false`
- `systemverilog.parallelProcessing > {total_number_of_files_in_workspace}/100`

joecrop, 2022-03-02:
OK, I found some issues with the regular expressions that parse port and parameter names. I found fixes for them and will incorporate them in the next release.

guojiazhuxi, 2022-03-04:
> OK, I found some issues with the regular expressions that parse port and parameter names. I found fixes for them and will incorporate them in the next release.

thank you！

joecrop, 2022-03-09:
This was just released in version 0.12.0

closed 2022-03-09 by joecrop

guojiazhuxi, 2022-11-26:
> 这是刚刚在版本 0.12.0 中发布的
When there is import, the input and output signal cannot be jumped, and the definition cannot be found

[image]