eirikpre/VSCode-SystemVerilog #139: Auto instantiation not reliable

PiyushSaini09, 2021-12-28:
Hi-
The auto instantiation feature seems unreliable. I'm using v0.11.3. Sometimes the feature automatically instantiates the module as expected and sometimes it just does not.  
In some scenarios it throws something like this - 
[image]

The module was instantiated but without the port list.
I cant figure out the issue.

I'm also getting this error in log-
[image]

joecrop, 2022-01-12:
I think I found the problem you are seeing. Are you using a Windows OS?

PiyushSaini09, 2022-01-12:
Yes. I am on windows 10.

joecrop, 2022-01-12:
OK, I just got a test suite up and running in windows, and I am seeing many issues with the module instantiator. It appears that the extension has only been tested on mac/liunx. I'll see if I can root cause it.

PiyushSaini09, 2022-01-13:
Understood. I'll close the issue then.

joecrop, 2022-01-13:
I think it is worth keeping this issue open until it is resolved, so others can find it if they have the same problem. I'll change the title to match our new understanding of the issue.

joecrop, 2022-01-16:
OK, I have found the cause of the problem. The extension works fine for me on Windows when compiled using Note v12. But the default Windows compile version seems to use Node v16. So we will have to get a grasp on why the code behaves differently on Node 16.

joecrop, 2022-01-16:
Well, more confusing details here. I took my Windows machine and downgraded back to Node 12, and now the tests are failing on node 12. I am getting a hunch that it is related to some node dependency installed by Chocolatey... but it is proving very difficult to pinpoint. Furthermore, I tried compiling using Mode 16 on Mac and Linux, both passed. So this still seems to be elusively contained within windows, and is not explicitly or directly related to the Node version.

joecrop, 2022-01-20:
OK, not the node version, and now I have confirmed that it is not the chocolaty version. I see some tests failing in Windows, but I think they are just file path related (i.e. `C:\blah` vs `/blah`) and not core function related.

@PiyushSaini09 A couple questions:

- Can you tell me more about your system? OS version, vscode version, extension version?
- Do you have other vscode extensions installed? Can you try disabling them?

I tried to recreate the problem you show in your screenshot but I can't get it to fail on Windows.

PiyushSaini09, 2022-01-21:
Just realized, I had another extension installed ([this one])(https://github.com/al8/sublimetext-Verilog). That's why the instantiation happened via that extension. Really sorry to mislead.

closed 2022-01-21 by joecrop