eirikpre/VSCode-SystemVerilog #142: `ifdef/`else/`endif folding

eng-tp, 2022-02-23:
Hi,

I'm seeing issues with folding whenever `ifdef/`endif pairs have `else or `elsif directives in between them.
These issues persist even when defining the foldingStrategy as indentation.
E.g.:

`ifndef COMP_GUARD
`define COMP_GUARD
  class x
    function z
      `ifdef A
        behavior 1
      `else //A
        behavior 2
      `endif //A
    endfunction
  endclass
`endif //COMP_GUARD

  - folding on `ifdef A folds code all the way to `endif //COMP_GUARD, ignoring the middle `else //A and `endif //A
  - folding on `else //A folds to `endif //A as expected

Looking at the languageSettings.json file for systemverilog, I see that:
  1. `else and `elsif are present in both start and end markers
  2. markers for all `ifdef - related directives do not assume \\b at the beginning (so, no indentation is detected and/or preserved)

I can't say whether any of these is responsible for the indentation failure - but I assume it's the latter -, and I also don't know how I can experiment with this locally: I am a new user to VSCode and its extensions, and it seems I can't edit the languageSettings.json file for the systemverilog extension locally - or I don't know how to do it...

This seems related with #65 , but that issue was closed and this behavior persists.

Can you please help me in figuring this out?

Thanks, and BR

joecrop, 2022-03-09:
Hi @eng-tp, I dug into this, and it appears to be a limitation of the vscode language configuration. You are right, `language-configuration.json` is the file that controls folding of ifdefs. Unfortunately, because `else` is both a begin and end pattern, vscode seems to be greedy and ignore it as an end and skips to `endif`.

It seems like we would need to implement a registerFoldingRangeProvider like [this extension](https://github.com/zokugun/vscode-explicit-folding/blob/master/src/extension.ts) does, as it supports `middle` expressions. But that is a pretty involved effort.

If you want to play around with this yourself, you can follow the [contributing](https://github.com/eirikpre/VSCode-SystemVerilog#contributing) guide to download the git repo and run the extension in the debugger.

joecrop, 2022-03-09:
Another potential solution is to install [this extension](https://marketplace.visualstudio.com/items?itemName=zokugun.explicit-folding) and configure it to support `else` folding.

eng-tp, 2022-03-09:
Hi @joecrop,

Thanks for your comments. I will see if I can find the time to experiment with the git repo.

In any case, to me a quick & dirty fix would be to remove the else/elsif from both the begin and end patterns, leaving just the ifdef/ifndef as begin, and endif as end - it would fold everything inside the same ifdef/endif, but it would be much better than the current behavior.

If I can try this out and get it to work, I'll let you know.

Cheers,
Tiago