eirikpre/VSCode-SystemVerilog #144: systemverilog.Indexingexclude not filtering directories

sarki94g, 2022-03-02:
I have the following line in my workspace settings:

"systemverilog.excludeIndexing": " \*/design/[!r]\*, \*\*/gate_level/\*\* ",
        "systemverilog.includeIndexing": [
          "\*\*/\*.{sv,v,svh, f}"
        ]

Yet when I select "Go to Definition" the module list I am present a list of 30 definitions of which most are in directories that should have been excluded.

[image]

joecrop, 2022-03-03:
I was looking through this code the other day and I think these variables
are not implemented correctly. I’ll look into if for the next release.

joecrop, 2022-03-08:
@sarki94g, I think your settings are correct. The extension caches previously parsed files. I wonder if the extension is just showing you parsed files from the cache. Can you try deleting the cache? It's a bit hard to find, but will be somewhere in here: `.config/Code/User/workspaceStorage` Alternatively, you can rename the workspace folder on the disk.

joecrop, 2022-03-12:
Actually, an easier way to reparse the workspace is to click the SystemVerilog extension button at the bottom of vscode.

[image]

joecrop, 2022-04-26:
@sarki94g changing the settings does not cause re-indexing as indexing can be an expensive task depending on the size of the workspace.

TCWORLD, 2023-10-13:
I've also been having issues with the file exclusion. It seems like the parsing of the setting doesn't split on commas when there are multiple files.

For example `"**/folder1"` works, `"**/folder2"` works, but  `"**/folder1,**/folder2"` doesn't exclude anything.

joecrop, 2024-01-05:
> also been having issues with the file exclusion. It seems like the parsing of the setting doesn't split on commas when there are multiple files.

Right, commas don't work the way you have written them. You have to use a properly formatted glob pattern. Do this instead: `**/{folder1,folder2}`
There is a handy glob test tool you can use here: https://www.digitalocean.com/community/tools/glob

joecrop, 2024-01-05:
> Does a change to the settings automatically re-index as happens with the Workspace directory browser?

No

closed 2024-01-05 by joecrop