eirikpre/VSCode-SystemVerilog #156: ${workspaceFolder} not resolving properly in settings.json

vjalle, 2022-04-17:
This does not work:
    "systemverilog.formatCommand": "**${workspaceFolder}**/Tools/verible/verible-verilog-format --indentation_spaces=4 --expand_coverpoints=true",
Simply nothing happens when invoking formatting, apart from the notification mentioned below, that's always there.

This does not work, either:
    "systemverilog.formatCommand": "v:/OneDrive/Emb/Tools/verible/verible-verilog-format --flagfile **${workspaceFolder}**/.verible-format",
I get this in the output view:
**ERROR: Can't open flagfile /V:/OneDrive/Emb/.verible-format**
The file is there, I guess the problem is the / at the beginning (using it on Windows).

Furthermore, when there is no reference to ${workspaceFolder} in the configuration at all, formatting works, but I always get a notification:
[image]
Found no way to get rid of this warning, I have no idea what's triggering it. Interestingly, only one computer is doing this, no warnings on another one with the same settings.
The config was as below:
    "systemverilog.formatCommand": "v:/OneDrive/Emb/Tools/verible/verible-verilog-format  --indentation_spaces=4 --expand_coverpoints=true",

joecrop, 2022-04-26:
This error seems to be because the file you are currently editing is not inside your workspace. Here's the code that generates the warning you are seeing:

```typescript
// From src/providers/FormatProvider.ts
const currentDocumentUri = editor.document.uri;
let workspacePath = vscode.workspace.getWorkspaceFolder(currentDocumentUri);
if (!workspacePath) {
    const fallbackWorkspace = vscode.workspace.workspaceFolders[0];
    vscode.window.showWarningMessage(`Expanding \${workspaceFolder} to '${fallbackWorkspace.name}'.`);
    workspacePath = fallbackWorkspace;
}
```

A couple thoughts:
1. Do you have multiple workspaces open in the same vscode session?
2. Do you have a systemVerilog file open in vscode and is that file within your current workspace?

vjalle, 2022-04-27:
Thanks for looking into this.
I had only one workspace open. You're correct: if the Verilog file is in the workspace, or I just open it's folder with Code, there is no warning. If I drag the file into a Code instance that was opened with a different workspace, I get the warning. Don't know whether it's normal or not, it's OK for me now that I understand the reason.

Clang-format and the C/C++ plugin work in a bit different way. When formatting a file not in the workspace, there is no warning, but the file is formatted with default options, not with the ones specified in .clang-format in the workspace root. Quite annoying, not really useful feature. For Verible the options are provided in the command, so when it's working (with or without warnings), it's always working well. I'd say, the logical behavior would be using the workspace settings even for files not in the workspace, and not throwing any warnings. Might be incompatible with the philosophy of Code though.

Not resolving ${workspaceFolder} in the command (even when working on files inside the workspace) seems to be a common issue affecting many (all?) plugins. Might be a limitation of Code itself? Would be good to fix this.

joecrop, 2022-05-02:
Yes, I think this is a vscode limitation. It looks at the file path to determine the `${workspaceFolder}` value. I think the plugin is actually doing what we expect here. It cna't find the correct workspace, so it defaults to workspace[0] and fires a warning. I think there are two actions for this issue:

1. Make sure paths are expanded correctly in windows
2. Change the warning to say something like: "<filename> is not found to be within a workspace, so the `wsName` workspace will be used for formatting settings.

vjalle, 2022-05-06:
I'd say the warning is not needed. All other workspace settings are applied anyway without warning. It's quite unusual, no other tooling complains when working with files not in the workspace.
Note the behavior of clang-format mentioned above. I can't imagine any use case for that. On the other hand, the opposite is useful: opening a foreign file and formatting it with my favorite settings.

joecrop, 2022-05-09:
OK, I'll move the warning to the output channel. That way it is still accessible, but hopefully not a burden.

PR #160 (2022-05-09): modified formatter command parser to support windows paths
Closes #156
closed 2022-05-09 by joecrop