eirikpre/VSCode-SystemVerilog #157: `systemverilog.includeIndexing` cannot resolve relative path

xiaosiyu0603, 2022-04-25:
Assuming my project is at `/home/user/project` with a subfolder `rtl`, and I only want to index files in `/home/user/project/rtl`, I open VS Code at `/home/user/project`. 
When I set "systemverilog.includeIndexing" as `rtl/**/*.{sv,v,svh,vh}` or `./rtl/**/*.{sv,v,svh,vh}` or `${workspaceFolder}/rtl/**/*.{sv,v,svh,vh}`, no files would be indexed. Only when it was set as `/home/user/project/rtl/**/*.{sv,v,svh,vh}` could the files be indexed.

My environment:
development via Remote-ssh
Local OS: Windows 7 (Windows_NT x64 6.1.7601)
Remote OS: Ubuntu 20.04 LTS (Linux version 5.4.0-107)
VS Code: 1.63.2 (both OS)
SystemVerilog - Language Support: 0.12.1

Would anyone know if there is any solution to solve the problem? Tks!

joecrop, 2022-04-26:
Yup, this is a bug. To work around this for now, please add a `*` as the first character of the setting. For example: `*rtl/**/*.{sv,v,svh,vh}`. This will switch the parser into a different mode that will make this work. I'll add a fix to search for relative files soon.

yiancar, 2022-04-26:
Hello an additional question.
I am using the remote ssh plugin.
From where I opened my folder, my rtl files are in `/rtl` and down, similar to above they can be in `/rtl/xyz`
is it reasonable to expect to use `**/rtl/**/*.{sv,v,svh,vh}` as well?

I have added
```
    "systemverilog.includeIndexing": [
        "**/logical/**/*.{sv,v,svh,vh}"
    ]
```
To my remote settings.

Now when I click the indexing button at the bottom it seems to be indexing for a very long time.

Given the project is VERY large I dont know how long it should take (some files are more than 4000 lines, and had to extend the limit)
Is there a way to see if its doing the right thing?

I tried to find the definition of a module but it cant find any :/

xiaosiyu0603, 2022-04-26:
@joecrop Thanks a lot! I followed your advice and it works well!

PR #158 (2022-04-26): Include indexing docs
Improved documentation on includeIndexing setting. Closes #157