eirikpre/VSCode-SystemVerilog #165: extension required many CPU used in TOP

huanghx1990, 2022-05-23:
Hi, I am using this extension, But in my login centos server, my collegues always notice me that my vscode consume CPU greatly, I checked use ps -aux and match the jobID showed in TOP and all of them are command just as below
{VSCODE_PATH}/resources/app/extensions/json-language-features/sever/dist/node/jsonServerMain --node-ipc --clientProcessId=xxx (each large CPU utility reach 100%), But I only open 1 vscode GUI. 
So I want to know what trigger this node-pic thread and how to control them in one GUI?

Best,
Xu

eirikpre, 2022-05-27:
With the 0.13.2 version we fixed a critical bug that would cause the CPU to hang.

Can you check if you are using the newest version?

Otherwise there are some settings to limit the extension's resource hunger

huanghx1990, 2022-07-03:
HI I check out the latest version and using it. I could not figure out obvious reduce of CPU utility. So I hungry wait for a newer version of this extension since it is big help for me to code or analyze systemverilog environment.

Bests,
Xu

joecrop, 2024-08-30:
Is this still happening for you? The `systemverilog.maxLineCountIndexing` should help you dial in the performance by excluding large files from being fully indexed.