eirikpre/VSCode-SystemVerilog #166: Extension causes JS heap allocation failure

emroch, 2022-05-25:
Issue Type: <b>Bug</b>

I have an SSH Remote workspace with many SV files in it. When enabled, this extension begins acquiring memory and eventually faults when the JS heap size is exceeded. I only have 3 SV files open in the workspace, but there are hundreds if not thousands in the whole workspace. The extension did not crash until today, though I have not been monitoring its memory usage so it is possible it was always using a lot and only now is bumping up against the limit. The remote machine I am running on has 8GB ram and the extension host uses about 2.2GB before crashing.

Extension version: 0.13.2
VS Code version: Code 1.67.2 (c3511e6c69bb39013c4a4b7b9566ec1ca73fc4d5, 2022-05-17T18:20:57.384Z)
OS version: Darwin x64 21.3.0
Restricted Mode: No
Connection to 'ssh-remote+aus-eroch01' could not be established

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i9-9880H CPU @ 2.30GHz (16 x 2300)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>metal: disabled_off<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: enabled<br>video_encode: enabled<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|3, 3, 2|
|Memory (System)|16.00GB (0.72GB free)|
|Process Argv|--crash-reporter-id 31303ca5-7d04-4a93-b6ab-f413410a38ff|
|Screen Reader|no|
|VM|0%|

Connection to 'ssh-remote+aus-eroch01' could not be established
</details>

joecrop, 2022-05-27:
Hi @emroch have you tried adjusting the settings based on the [recommendations](https://github.com/eirikpre/VSCode-SystemVerilog/blob/master/README.md#recommendations) section of the readme?

apippin, 2022-08-12:
I'm seeing the same error. I have tried adjusting the settings based on the recommendations section. We need a way to increase the heap memory size to enable us to use this extension on very large designs.